HesaiTechnology/HesaiLidar_SDK_2.0 #4: pcap文件转为pcd文件

bingo-gby, 2023-12-08:
你好，我想问一下，是否有现成的包支持把pcap文件，解析成pcd文件，我用的你们的pandarview2软件貌似 dump出来的pcd文件有问题......

bingo-gby, 2023-12-08:
> 需要打开PCD的宏，这里的参数配置需要改的是PCAP路径，和firetime路径，firetime选择你用的雷达型号

明白，那我应该pcl_tool 和 test.cc都改了，这是ok的吧？

bingo-gby, 2023-12-08:
我把pcl_tool 和 test.cc 的数据来源和一些配置都改了，但是最后好像读取还是有问题
[image]
请问这是啥原因呢

GIS-PanTianLi, 2024-09-08:
> 我把pcl_tool 和 test.cc 的数据来源和一些配置都改了，但是最后好像读取还是有问题 [...]

请问你解决了吗

bingo-gby, 2024-09-09:
没有解决，后面是用的软件把pcap包转为pcd文件，然后进行resize

Flesymeb, 2025-03-04:
> 
> 我把pcl_tool 和 test.cc 的数据来源和一些配置都改了，但是最后好像读取还是有问题  请问这是啥原因呢
> 
> 请问你解决了吗

您好，请问使用了什么软件将pcap批量转为pcd呢，我用SDK也没解析出

changxingshuo, 2025-04-17:
修改pcl_tool.cc文件即可，首先选择你需要的文件格式，例如下述开启了第一种格式
```
#define SAVE_PCD_FILE_ASCII
// #define SAVE_PCD_FILE_BIN
// #define SAVE_PCD_FILE_BIN_COMPRESSED
// #define SAVE_PLY_FILE
```
然后再main函数中修改对应参数，如果是解析pcap，将source_type修改为DATA_FROM_PCAP，在pcap_path中填写你的pcap路径，correction_file_path中填写角度校准文件路径，如果需要也要填写firetimes_path对应的firetimes校准文件路径。然后运行就可以了。

changxingshuo, 2025-04-17:
Packet with invaild delimiter是雷达初始化时，收到了非点云包数据的警告。后续正常运行代表初始化成功，该警告忽略。

closed 2025-04-17 by changxingshuo