HesaiTechnology/HesaiLidar_SDK_2.0 #12: PCAP转PCL失败

GIS-PanTianLi, 2024-09-07:
[image]
请问为什么是这种情况？

HesaiYR, 2024-09-23:
你好，这个看起来是可视化VTK库相关警告信息，但应该不会影响pcap解析为pcl。另外，从最后一句话的打印可以看出，SDK已经正常解析了点云数据。请问可以再详细描述下你这边遇到的问题吗？

GIS-PanTianLi, 2024-09-23:
> 你好，这个看起来是可视化VTK库相关警告信息，但应该不会影响pcap解析为pcl。另外，从最后一句话的打印可以看出，SDK已经正常解析了点云数据。请问可以再详细描述下你这边遇到的问题吗？

Packet with invalid delimeter.

HesaiYR, 2024-09-23:
> > 你好，这个看起来是可视化VTK库相关警告信息，但应该不会影响pcap解析为pcl。另外，从最后一句话的打印可以看出，SDK已经正常解析了点云数据。请问可以再详细描述下你这边遇到的问题吗？
> 
> Packet with invalid delimeter.

这个是因为驱动收到的数据包并不是0xeeff开头。麻烦确认下雷达使用的型号，或者录制一段pcap数据发我们解析下。请发至：https://www.hesaitech.com/cn/support
[image]

closed 2025-04-17 by changxingshuo