HesaiTechnology/HesaiLidar_SDK_2.0 #21: Does HesaiLidar_SDK_2.0 and HesaiLidar_General_SDK have different manuals

robotichustle, 2025-04-01:
Hi, I need to get my Pandar XT32M2X lidar working with ROS2. I noticed that the ROS2 support in HesaiLidar_General_ROS (and HesaiLidar_General_SDK) is incomplete / no longer in development. But I see that I should be moving to the newer HesaiLidar_ROS_2.0 repository (based on HesaiLidar_SDK_2.0) where ROS2 support is actively being developed. I had a few customisations on the old driver that I want to migrate across to ROS2. I had to understand the XT32M2X User Manual for packet structure details. Now to do the migration, is this old user manual sufficient for HesaiLidar_SDK_2.0 or has the packet structures for this lidar changed - which means I would need to request a new one that corresponds to this new SDK 2.0 version? Looking forward to your clarification.

kExU9853, 2025-04-01:
kind of the same issue, waiting for the reply

changxingshuo, 2025-04-17:
The packet structure of the Pandar XT32M2X Lidar has not changed and can continue to be used.
The structure is defined in udp_protocol_v6_1.h and the parsing process is in udp6_1_parser.cc

closed 2025-04-17 by changxingshuo