HesaiTechnology/HesaiLidar_SDK_2.0 #23: sdk.stop()

St3ff3nBusch, 2025-04-10:
# Stop Bug
Hello,
 please help me fixng the bug in the sdk Stop() method.

I use the SDK to handle the life cycle of the lidar: configure, activate, deactivate and shutdown.
The programm is starting by :

```
sdk.Init(params); //for the configuration
```
The reconfigure state is working fine with the first call and use : 
```
sdk_.Stop();
sdk_.Init(param_);
sdk_.lidar_ptr_->SetSpinSpeed(sdk_.lidar_ptr_->ptc_client_, current_lidar_speed_);
setOperatingMode(1);
sdk_.Start();
```
Activate and deactivat works fine with :
```
sdk_.lidar_ptr_->ptc_client_->SetStandbyMode(standby_mode)) 
```

Shutdwon is just using :
```
sdk_.Stop();
```

Shutdown and a second configure call is killing the program.
 ## Error Messages:
release Udp6_1parser
release general Parser
TcpClient::Close()
SocketSource::Close()
Source::Close()
terminate called after throwing an instance of 'std::system_error'

changxingshuo, 2025-04-17:
‘sample.Stop();’ should be fine, the problem is after this function.
test.cc uses ‘IsPlayEnded(sample)’ after ‘sample.Stop();’, which calls the lidar_ptr_ pointer inside, but ‘sample.Stop();’ already frees it, so it will lead to core dumped.

St3ff3nBusch, 2025-04-23:
Hi Changxingshou,

But I don't use either the test.cc or IsPlayEmbeded(sample).
I register a callback :
 ```
sdk_.RegRecvCallback(
      [this](const LidarDecodedFrame<LidarPointXYZIRT> &frame) {
        processLidarFrame(frame);
      });
 ```
but the callback starts with an debug output, which isn't reached after stop().

Regards 
Steffen

changxingshuo, 2025-04-24:
I observe that you have registered the callback function for LidarDecodedFrame, which is called during the parsing process, specifically after the point cloud is framed.
When the parsing process is started using the start() function, this callback function is triggered after a certain period of time (typically 10hz) if the correct point cloud is received.
But when you use stop(), it will stop the parsing process, then this callback function will not be called anymore.

changxingshuo, 2025-04-24:
The main thing to do after calling stop() is to stop using the pointer variables in sdk_, as they will all be freed and will cause a core dump if used directly.

closed 2026-11-02 by ForStack