HesaiTechnology/HesaiLidar_SDK_2.0 #33: sdk解析pcap数据卡顿

natsume7, 2025-05-30:
对Pandar40P获取的pcap数据使用sdk进行解析时，会出现卡顿现象（pandarview显示有201帧，sdk解析到141帧卡顿3分钟左右后会再继续解析），如何解决该问题

[image]

changxingshuo, 2025-05-30:
pcap_play_synchronization默认为true，会获取点云中的时间，做差值，然后与本地消耗的时间作比较，如果本地消耗的时间较少，会等待一段时间，目的是模拟PCAP中点云的实际时间的前进过程。  141帧和142帧的点云时间之间，应该出现了3分钟左右的跳变，因此触发了等待3分钟。
可以将pcap_play_synchronization置为false即可

changxingshuo, 2025-06-03:
1. 如果使用ROS+SDK，在ROS的config.yaml中的lidar下的driver下的pcap_play_synchronization属性改为false
2. 如果是使用SDK的test.cc的话，在配置参数的地方，在main函数中添加param.decoder_param.pcap_play_synchronization = false;

natsume7, 2025-06-03:
> 1. 如果使用ROS+SDK，在ROS的config.yaml中的lidar下的driver下的pcap_play_synchronization属性改为false
> 2. 如果是使用SDK的test.cc的话，在配置参数的地方，在main函数中添加param.decoder_param.pcap_play_synchronization = false;

在libhesai/driver_param.h找到了该属性修改后可以运行，但还存在最后一帧不会解析（pandarview里显示的是0~200，只解析到了199，对于其他pcap测试解析时情况亦如此）、不会退出while循环打印printf("The PCAP file has been parsed and we will exit the program.\n");测试使用的是[https://github.com/HesaiTechnology/HesaiLidar_SDK_2.0/blob/master/test/test.cc屏蔽了std::this_thread::sleep_for(std::chrono::milliseconds(1000))]

changxingshuo, 2025-06-05:
@natsume7 需要修改一下代码，比如hesai_lidar_sdk.hpp的Run()函数中对GetOnePacket返回-1时，加一个判断，如果持续一段时间(例如100ms)没有获取到数据，且packet_index大于0（有数据没有被推出）。 则触发一次推数据的逻辑（触发point_cloud_cb_）。

natsume7, 2025-06-05:
> [@natsume7](https://github.com/natsume7) 需要修改一下代码，比如hesai_lidar_sdk.hpp的Run()函数中对GetOnePacket返回-1时，加一个判断，如果持续一段时间(例如100ms)没有获取到数据，且packet_index大于0（有数据没有被推出）。 则触发一次推数据的逻辑（触发point_cloud_cb_）。

具体如何修改，可否在hesai_lidar_sdk.hpp代码处示例修改一下

NbbZm, 2025-06-05:
用sdk2.0在windows下进行离线pcap解析非常慢，每帧平均在300ms以上，这个是什么原因，如何加速？使用的是PandarXT-16，劳烦请教一下

changxingshuo, 2025-06-06:
@natsume7 
      //get one packte from origin_packets_buffer_, which receive data from upd or pcap thread
      static int getEmpty = 0;
      int ret = lidar_ptr_->GetOnePacket(packet);
      if (ret == -1) {
        getEmpty++;
        if (getEmpty >= 10 && packet_index > 0) {
          while(!lidar_ptr_->ComputeXYZIComplete(packet_index)) std::this_thread::sleep_for(std::chrono::microseconds(100));
          if(point_cloud_cb_) point_cloud_cb_(lidar_ptr_->frame_);
          packet_index = 0;
          lidar_ptr_->frame_.Update();
        }
        continue;
      }
      getEmpty = 0;
修改如上

closed 2025-07-01 by changxingshuo