HesaiTechnology/HesaiLidar_SDK_2.0 #38: #pragma pack(push, 1) usage is wrong

tly000, 2025-06-06:
The SDK uses the following define at multiple locations: 

```cpp
#ifdef _MSC_VER
#define PACKED
#pragma pack(push, 1)
#else
#define PACKED __attribute__((packed))
```

This is wrong and leads to crashes. `#pragma pack(push, 1)` changes the alignment of all structs and classes that follow, this means it breaks the alignment for all classes like `std::vector<...>` or other standard types.

The correct usage is: 

```cpp
#pragma pack(push,1)
struct MyPackedStruct {
    ...
};
#pragma pack(pop)
```

This is also supported in GCC and clang, so the `PACKED` macro could be removed. Instead only `#pragma pack(push,1)` and `#pragma pack(pop)` could be used.

changxingshuo, 2025-06-09:
The following code is also currently used
`#ifdef _MSC_VER
#pragma pack(pop)
#endif`
Next version will remove `#define PACKED`

changxingshuo, 2025-07-01:
Updates completed