HesaiTechnology/HesaiLidar_SDK_2.0 #39: Several Source Files contain a GPL-2.0-or-later reference

AVL-OpenSourceOffice2, 2025-06-16:
The following source files 

"path": "HesaiLidar_SDK_2.0-master/libhesai/PtcClient/include/client_base.h",
"path": "HesaiLidar_SDK_2.0-master/libhesai/PtcClient/include/ptc_client.h",
"path": "HesaiLidar_SDK_2.0-master/libhesai/PtcClient/include/tcp_ssl_client.h",
"path": "HesaiLidar_SDK_2.0-master/libhesai/PtcClient/src/tcp_ssl_client.cc",
"path": "HesaiLidar_SDK_2.0-master/libhesai/SerialClient/include/serial_client.h",
"path": "HesaiLidar_SDK_2.0-master/libhesai/Source/include/serial_source.h",
"path": "HesaiLidar_SDK_2.0-master/libhesai/Source/src/serial_source.cc",

all contain the following GPL-2.0-or-later header

/*
 * Copyright (C) 2019 Hesai Tech<http://www.hesaitech.com>
 *
 * This program is free software; you can redistribute it and/or
 * modify it under the terms of the GNU General Public License
 * as published by the Free Software Foundation; either version 2
 * of the License, or (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA  02111-1307, USA.
 */

In my understanding, this leads to **GPL-infection**. 
However, I can see that the IP-rights belong to  -> Hesai Tech<http://www.hesaitech.com>

Since I can't see and do not know how the FOSS steward of this repo is connected to this company here are my questions:

- is there a permission from the IP-owner to release these files under the BSD-3-Clause license?
- or is this actually an error in the stated license information for this component?  If so, the license information should look like this -> 
  BSD-3-Clause AND GPL-2.0-or-later

Kind regards
Bernhard

changxingshuo, 2025-06-25:
Sorry, we internally confronted this issue yesterday and believe that the probability is that this agreement is the result of a historical legacy.
We will complete this troubleshooting this week, and if not involved will remove and standardize on the 2023 version of the protocol.

AVL-OpenSourceOffice2, 2025-06-25:
Dear changxingshuo, 

thank you for your reply! 
We appreciate your efforts to fix this issue in a timely matter.

Many thanks in advance and 
Kind regards
Bernhard

changxingshuo, 2025-07-01:
Updates completed

closed 2025-07-01 by changxingshuo