HesaiTechnology/HesaiLidar_SDK_2.0 #44: Question about GetDistanceCorrection

fulkast, 2025-07-23:
https://github.com/HesaiTechnology/HesaiLidar_SDK_2.0/blob/64b36f9d0b2ecfb86da367088cbc71ec08286c3d/libhesai/UdpParser/src/general_parser.cc#L336

I am confused by the coordinate correction code above. It adds a constant optical center term for all points in the cloud. However, the optical center is in constant rotation with respect to the geometric center, and hence I believe that the optical center addition term should be different depending on the azimuth. How have I misunderstood this distance correction code?

changxingshuo, 2025-07-28:
1. The input to this correction function contains information about two angles, which are relative to the lens (transmit or receive). and is held variable for a given radar specified.
2. these two angles form a direction vector pointing in the direction of the light passing through, and likewise the direction of the object. Suppose that the lens is point S, the origin is point O, and the light reaches the object at point T. SOT is a triangle.
3. During the rotation, the relative positions of S and O remain the same, as does the direction of the ST vector. The object's proximity changes only the length of ST.
4. the initial calculation is the coordinate of T relative to S. The function is corrected by correcting the point S to the point O and calculating the coordinate of T relative to the point O. The function is then modified to calculate the coordinate of T relative to the point O. The relative position of the triangle SOT does not change during the selection process, so the coordinate correction is independent of the rotation itself.

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

changxingshuo, 2025-07-29:
1. These two points rotate. But there are several points whose relative positions are constant.
2. there are three angle information in the LiDAR, the first one is the horizontal angle of the LiDAR, we can think that this horizontal angle is rotating from 0 to 360 degrees. The second is the horizontal angle in the angle calibration file, and the third is the vertical angle in the angle calibration file. The last two are fixed values.

[image]

3. As shown in the figure, when Azi is determined, the above two angles are also determined, i.e., the direction of ST, and the position of point S with respect to point O has also been determined. Assuming that the distance of the object T remains constant at 360 degrees, the triangle SOT is fixed.
4. when the radar is rotating, i.e. Azi is rotating, point S is moving along the circle in the figure. It can be seen as the triangle SOT is constantly rotating. Even if the distance of T changes, it is only a change in the length of ST and does not affect the position of SO.
5. Therefore, no matter how the emitter and receiver rotate, when Azi is fixed, the triangle SOT is also fixed and is not affected by Azi. That is, it is not affected by the rotation factor.

closed 2026-10-27 by ForStack