HesaiTechnology/HesaiLidar_SDK_2.0 #50: pcap转pcd

raymondwm, 2025-09-08:
https://github.com/HesaiTechnology/HesaiLidar_SDK_2.0/blob/master/docs/save_point_cloud_data_as_a_pcd_file_CN.md
pcap转pcd 没有成功，可能是什么问题？
```
#ifdef PCAP_PARSER_TEST
  param.input_param.source_type = DATA_FROM_PCAP;
  param.input_param.pcap_path = "/mnt/20250626161519_F.pcap";
  param.input_param.correction_file_path = "Your correction file path";
  param.input_param.firetimes_path = "Your firetime file path";

  param.decoder_param.pcap_play_synchronization = true;
  param.decoder_param.pcap_play_in_loop = false; // pcap palyback
```

root@cce8b29242ad:/mnt/HesaiLidar_SDK_2.0/tool/build# pwd
/mnt/HesaiLidar_SDK_2.0/tool/build
root@cce8b29242ad:/mnt/HesaiLidar_SDK_2.0/tool/build# ls
CMakeCache.txt  CMakeFiles  Hesai_lidar_sdk  Makefile  cmake_install.cmake  packet_loss_tool  pcl_tool
root@cce8b29242ad:/mnt/HesaiLidar_SDK_2.0/tool/build# ./pcl_tool
-------- Hesai Lidar SDK V2.0.10 --------
[2025-09-08 02:55:17][INFO]logger start to run
[2025-09-08 02:55:17][INFO]Use thread num 1 to parse packet
[2025-09-08 02:55:17][INFO]Lidar::ParserThread start to run
[2025-09-08 02:55:17][INFO]set thread 281472912089312, tid 17941, policy 1 and priority 70
[2025-09-08 02:55:17][INFO]get thead 281472912089312, tid 17941, policy 0 and priority 0
[2025-09-08 02:55:17][INFO]Lidar::Receive Udp Thread start to run
[2025-09-08 02:55:17][INFO]set thread 281472920482016, tid 17940, policy 1 and priority 70
[2025-09-08 02:55:17][INFO]get thead 281472920482016, tid 17940, policy 0 and priority 0
[2025-09-08 02:55:17][WARNING]can not parser Ethernet data, type = f788 
[2025-09-08 02:55:17][WARNING]can not parser Ethernet data, type = f788 
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter
[2025-09-08 02:55:17][WARNING]Packet with invaild delimiter

raymondwm, 2025-09-08:
想要解析离线文件pcap为pcd，按照文档设置了，运行时提示以下信息是正确的吗？要等这些信息不刷屏就算解析完了吗？？

这里有一个Lidar::Receive Udp Thread start to run   是不是误导？？

root@3bb57c227e26:/mnt/HesaiLidar_SDK_2.0/tool/build# ./pcl_tool 
-------- Hesai Lidar SDK V2.0.10 --------
[2025-09-08 11:50:37][INFO]logger start to run
[2025-09-08 11:50:37][INFO]Use thread num 1 to parse packet
[2025-09-08 11:50:37][INFO]Lidar::Receive Udp Thread start to run
[2025-09-08 11:50:37][INFO]Lidar::ParserThread start to run
[2025-09-08 11:50:37][INFO]set thread 281472476893408, tid 480, policy 1 and priority 70
[2025-09-08 11:50:37][INFO]get thead 281472476893408, tid 480, policy 0 and priority 0
[2025-09-08 11:50:37][INFO]set thread 281472485286112, tid 479, policy 1 and priority 70
[2025-09-08 11:50:37][INFO]get thead 281472485286112, tid 479, policy 0 and priority 0
[2025-09-08 11:50:37][WARNING]can not parser Ethernet data, type = f788 
[2025-09-08 11:50:37][WARNING]can not parser Ethernet data, type = f788 
[2025-09-08 11:50:37][WARNING]Packet with invaild delimiter
[2025-09-08 11:50:37][WARNING]Packet with invaild delimiter
[2025-09-08 11:50:37][WARNING]Packet with invaild delimiter

changxingshuo, 2025-10-30:
@UP-programmer correction_file_path需要设置一个本地的角度文件。
将其他宏禁用，要开启PCAP_PARSER_TEST这个宏

raymondwm, 2025-10-31:
> 你解析成功了吗

成功了，文件的问题，换正确的文件就行。

UP-programmer, 2025-10-31:
能加个联系的方式吗，我好像配环境卡住了，qq2065225637

         原始邮件

发件时间：2025年10月31日 15:28
收件人：HesaiTechnology/HesaiLidar_SDK_2.0 ***@***.***&gt;
抄送：UP-programmer ***@***.***&gt;, Mention ***@***.***&gt;

UP-programmer, 2025-10-31:
编译的时候报错，这个怎么解决

[image]

changxingshuo, 2025-11-05:
@UP-programmer window编译请参照readme中的编译说明。 且建议从git上拉代码，否则会因为utf标准问题导致识别错误